meshtastic/web-flasher #21: Web Flasher could not flash board,  http://flasher.pdxlocs.com did first go

Watto2021, 2024-01-19:
Hi have these board's
https://www.aliexpress.com/item/1005005967763162.html?spm=a2g0o.order_list.order_list_main.5.54bd1802PaOH9l 

[image]

from Ali express (Tenstar Robot) SX1276 ESP32 LoRa 'Heltec V2 screen printing" and trying to flash with web installer with no success. I tried thru py no success either. With esphome I could flash an incorrect version firmware to board but no Oled working, Could not flash the desired chosen firmware Heltec V2 2.17.dbac2b1 beta.
I tried a sketch i had for Arduino with SSD1306 Oled driver and flashed successfully and proved Oled was functional.
After much searching I seen mention of an "Old" no longer supported Installer as a possible backup (http://flasher.pdxlocs.com) . I connected my 3 boards to this and selected Heltec V2.1 and latest beta firmware which the number matches what I had already downloaded and attempted to flash. 
This Installer Successfully flashed all 3 boards first try with the desired firmware AND the Oled screen is now functional.
Is there an issue with the Newest recommended Web Installer?

garthvh, 2024-01-19:
The heltec v2 is not a supported model, you will need to flash it with the CLI

Watto2021, 2024-01-19:
> The heltec v2 is not a supported model, you will need to flash it with the CLI

yes I seen that on mesh website, tried thru py cli and thru windows command prompt to upload same firmware and does not work.
The boards are working now with Meshtastic flashed and connecting to android and PC thru arduino IDE.
My question is Why the Old Web Installer works fine yet the Newer Web Installer does not. As well as my experience here for anyone else trying to flash these boards unsuccessfully multiple different ways

closed 2024-01-30 by thebentern