meshtastic/web-flasher #26: Project works on localhost but doesnt when deployed to vercel

its-aomi, 2024-01-30:
Hi the project seems to work fine on localhost however on deployment to vercel it doesn't. Looks like due to the api that gets the firmware list as everything works fine except for the firmware list which doesn't show when deployed to vercel. 

How do i overcome this and load the firmware list?
[image]

sachaw, 2024-01-30:
Yeah, that's the CORS policy we have, is there any reason you want to deploy this yourself?

its-aomi, 2024-02-01:
Yeah, i am trying to deploy this to my own vercel account for testing purposes, tried on localhost worked fine, but tried online and didnt. Is there any way i can host the api myself?

sachaw, 2024-02-01:
Yes, you can spin up an instance of the API.
It is located in the meshtastic/api repo.

closed 2024-02-06 by thebentern