meshtastic/web-flasher #63: Alpha 2.3.5.2 on t-echo has issues with double click

austin20202020, 2024-04-17:
t-echo freezes up, I believe its happening when you double click the bottom button and it is the GPS settings causing the issue. 

If you have the internal gps set to disabled and use your phone gps then when you double click the bottom button to send an ad-hoc ping, it instead says on the e-ink screen "gps disabled", and then after that it freezes up.

If I have it set up ise the internal gps then the device works fine and double click does send an ad-hoc ping as expected.

todd-herbert, 2024-04-17:
There is no 2.3.5 alpha release as far as I am aware, do you mean that you have built the firmware from source, or are you using a different version?

I have just tested now, combining "Provide phone location to mesh" with GPS ENABLED, DISABLED, and NOT_PRESENT. 
I cannot reproduce the issue.

My first suggestion is to erase your flash, re-upload the firmware, then see if everything is normal. You can do this from http://flasher.meshtastic.org

[image]

In future, issues like this might be more suited for https://github.com/meshtastic/firmware/

austin20202020, 2024-04-17:
sorry, I realize now I posted in the wrong place. I'll test further tomorrow

closed 2024-04-17 by austin20202020