meshtastic/web-flasher #64: [Request] Update device list to be categoriesed by Vendor?

Talie5in, 2024-04-23:
The current list is getting long and requires scrolling to get down to beyond Nano G1 (at 1920x1200) so probably worse at 1920x1080.
Is it worth doing sub-tree's based on Hardware Vendor, with the ability to search to filter down the list?)

Even if done for the Vendors with many options and leaving the rest in the maintree if they only have 1 deivce (eg, CanaryOne, Hydra, unPhone)

Would make the WebFlasher UX a little simpler/easier to navigate.
eg, 
[FILTERBOX]
Lilygo -> 
------- T-Beam
------- T-Deck
------- T-Beam
------- T-Beam S3 Core
Heltec -> 
------- V3 
------- Wireless Paper 
------- Wireless Paper V1.0
------- Wireless Tracker
rPi ->
------- Pico 
------- Pico W
------- Pi Computer S3
RakWireless ->
------- RAK11310
------- RAK4631
CanaryOne
Hydra
Unphone

thebentern, 2024-06-23:
Ended up doing this by architecture #64 for now, but may revisit

closed 2024-06-23 by thebentern

Talie5in, 2024-06-23:
@thebentern Looks smexy like that, just noting however rak11310 ended up under RPi Pico banner.

However still a long list initially, worth starting with them collapsed?

Hopefully will revisit, I think by Vendor may be more intuitive to most end users.