meshtastic/web-flasher #68: Pico W detected as a Pico

clwgh, 2024-05-04:
### Summary

Connecting a Pico W-based node to the web flasher and selecting the automatic selection, it incorrectly detects it as a Pico, not a Pico W, which I believe uses different firmware.

### Select the auto option, connection prompt
[image]

### Pico erroneously detected
[image]

### Pico W is indeed available to select
[image]

thebentern, 2024-05-04:
The auto-detect returns whatever the hardware model protobuf the device has on it. It's an imperfect system because they are sometimes re-used between different hardware variants like the pico and a couple of the diy esp32 boards. Sadly I have no way of fixing this problem in the web flasher.

closed 2024-05-04 by thebentern

clwgh, 2024-05-04:
I believe that the firmwares for the Pico and Pico W are different and not entirely compatible. Is that correct? If so, it would be worth preventing the auto detect from selecting either the Pico or Pico W and instead directing the individual to make the correct manual selection. This will prevent incompatible firmware from being applied.

thebentern, 2024-05-05:
Indeed. I'm wondering if we can highlight the potential matched ones in the list

> Edit - to put it another way, restrict the auto-select to those models for which there is no such ambiguity. For all other models, the detection pops up a notice to the user asking them to make the correct selection manually.