meshtastic/web-flasher #70: Possible issues upload release zip

todd-herbert, 2024-05-20:
Full disclosure: I'm not entirely sure that I'm using this feature correctly.

What I'm doing:
* Visit http://flasher.meshtastic.org (tested both Chrome and MS Edge, Windows 10)
* Select device (tested with Heltec Wireless Paper)
* Click folder icon ("upload your own firmware release zip")
* Select file `D:\Downloads\firmware-2.3.9.f06c56a.zip`, downloaded from https://github.com/meshtastic/firmware/releases/download/v2.3.9.f06c56a/firmware-2.3.9.f06c56a.zip
* Press update
* Select the serial device

Process begins, then appears to hang:

```
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: dc:54:75:ce:bb:38
Uploading stub...
Running stub...
Stub running...
```

thebentern, 2024-05-20:
I can repro this. Seems like the matching regex for the update bin is borked for uploaded zips

thebentern, 2024-06-24:
Should be fixed with #77

closed 2024-06-24 by thebentern