meshtastic/web-flasher #81: flasher full erase option not presented when flashing a manually specified file

brianczapiga, 2024-08-09:
The full erase option is not presented when flashing a manually specified file.

This erase option only appears when selecting from the API driven list.

brianczapiga, 2024-08-16:
Also, when doing a full erase, the filesystem should probably be flashed again, perhaps there should be an advanced option to specify a file to be flashed for the filesystem.

thebentern, 2024-08-23:
> The full erase option is not presented when flashing a manually specified file.

This is somewhat by design, because doing a full install requires the filesystem, ble-ota, and application bins. Most users are installing the application bin update from build artifacts in Github.

brian-czapiga-okcupid, 2024-08-23:
Perhaps an advanced setting to allow specifying an offset to flash the bleota or littlefs at the either a manually specified offset or the appropriate offset based on the name of the file given.

brianczapiga, 2024-08-23:
Perhaps an advanced setting to allow specifying an offset to flash the bleota or littlefs at the either a manually specified offset or the appropriate offset based on the name of the file given?

thebentern, 2024-08-23:
That's kind of outside of the scope of this utility. At that point, you can just go to esptool.js's example and upload those bins and place the offsets to flash https://espressif.github.io/esptool-js/

One option where I allow custom file uploads to fully erase and flash is if you zip it and keep things in the structure of the official Github release firmware zips. You might try that if you're interested in the functionality with a little bit more ceremony.

closed 2024-08-23 by thebentern