meshtastic/web-flasher #105: Erase Flash Option w/ Custom zip or bin

medentem, 2024-10-02:
It appears when flashing a custom zip or bin file, the erase flash option is not available. That would be super helpful to properly make use of the `userprefs` capability which requires a wiped device.

thespillmonkey, 2024-10-06:
You need to use the firmware.factory.bin file from compile. And it needs to be named exactly that. 
And it will work :)

jp-bennett, 2024-10-07:
Trying this with my tbeam, both on the official flasher and a locally hosted one, I'm getting "invalid header: 0xa8120c42" spammed in the log afterwords. Is this something weird with my builds?

thebentern, 2024-10-07:
I'm assuming this needs to be the merged binary at 0x00.

medentem, 2024-10-07:
This was fixed with https://github.com/meshtastic/web-flasher/pull/106

closed 2024-10-07 by medentem