meshtastic/web-flasher #116: Placeholder: Setup partition scheme for larger flash devices and better handling of Device UI bins

thebentern, 2024-10-29:


mverch67, 2024-11-14:
Current 4MB partitioning:
```
# This is a layout for 4MB of flash
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x009000, 0x005000,
otadata,  data, ota,     0x00e000, 0x002000,
app,      app,  ota_0,   0x010000, 0x250000,
flashApp, app,  ota_1,   0x260000, 0x0A0000, <== bleota.bin
spiffs,   data, spiffs,  0x300000, 0x100000 <== littlefs.bin
```
8MB (Indicator):
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x330000,  <== currently 464kB free until littlefs gets in the way
app1,     app,  ota_1,   0x340000,0x330000,  <== bleota-s3.bin should go here
spiffs,   data, spiffs,  0x670000,0x180000,
coredump, data, coredump,0x7F0000,0x10000
```

16MB (T-Deck and others):
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,  <== currently 314kB free until littlefs gets in the way
app1,     app,  ota_1,   0x650000,0x640000,,  <== bleota-s3.bin should go here
spiffs,   data, spiffs,  0xc90000,0x360000,
coredump, data, coredump,0xFF0000,0x10000
```

mverch67, 2024-11-14:
Bundling the .cvs file with the firmware zip would allow a completely generic approach, that works for all future variations without the need for device type specific adaptations or other related maintenance. If no .csv included assume 4MB layout.

mverch67, 2024-11-15:
Alternative to csv files: use a manifest file, e.g. as described here:
https://github.com/esphome/esp-web-tools

[image]