meshtastic/web-flasher #130: Heltec T114 V2 corrupted filesystem after several weeks.

edegraaff, 2025-01-05:
Hi,  i have a Heltec t144 running as test router and the system had over the last months twice a corrupted filesystem. After reflashing it works again, sometimes i need to whipe the whole device before it accepts the new meshtastic image. I have no clue it is the mesthastic code causing this issue or the device itselves. The Tbeam, same version firmware of mesthastic, is stable like a rock. 

Logging attached, i will reflash with the latest beta and see what happens. 

Regards
eelco
[meshtastic-log-2025-01-05T12-52-18.040Z.log](https://github.com/user-attachments/files/18310292/meshtastic-log-2025-01-05T12-52-18.040Z.log)

b8b8, 2025-01-05:
This may be more appropriate for the "firmware" section. It also seems to be more prevalent on the all NRF devices. I have seen similar issues on the T1000E. What firmware version are you running?

edegraaff, 2025-01-05:
firmware-heltec-mesh-node-t114-2.5.14. was the one that was broken. I now was able to uploaded the official 2.4.2.7d00e1ce from the heltec site. Fall back to older releases that where on the download site of Heltec, i hope this is more stable.  I could move the discussion to the firmware location...

b8b8, 2025-01-05:
Id suggest moving to a 2.5+ version. I have been running 2.5.18 on my T114 and has been working ok so far.

edegraaff, 2025-01-05:
> Id suggest moving to a 2.5+ version. I have been running 2.5.18 on my T114 and has been working ok so far.
> 
> Anything previous to 2.5, is missing quite a few new features and is overall more detrimental to the mesh, as traffic was handled differently on those versions and was not optimized.

Ok, i will try to reflash, earlyer today the code was revoked and i was not able to download the latest version. Do you also run 24x7 for several weeks? The issue occurs after 4 to 8 weeks 24x7

b8b8, 2025-01-06:
For the one with this issue yes it was running for months, but it was a rak, not the T114. My T114s are handhelds that are not on all the time. I really think it is a general NRF issue though, not specific to the T114. 2.5.18 has been running well so far.

edegraaff, 2025-01-06:
When i entered the Meshtastic site yesterday and i wanted to download, the site showed a notice the firmware for the T114 was not availible. A few hours earlyer the firmware was there. Maybe there was a small mallfunction (firmware buildstreet active?) . It works now fine. I will upload the latest beta in the device. I also contacted Heltec, and they seem to have direct line with the dev team. Thanks for helping, and sharing thougths. 
Regards Eelco

peterhanneman, 2025-01-15:
I have four T114 v2 nodes running v2.5.18 will keep an eye on them but so far no issues.