meshtastic/web-flasher #144: Failed to open serial port.

LazyHatGuy, 2025-03-14:
On both Ungoogled-Chromium 134.0.9668.35 and on Microsoft Edge 134.0.3124.68 the following error is shown when clicking the rocket next to Select Target Device

ERROR	[iMeshDevice:SerialConnection]	Connect ❌ Failed to execute 'open' on 'SerialPort': Failed to open serial port.

The flasher also fails to flash or reset a LILYGO-LoRa T3-S3 on both browsers

ThatKalle, 2025-03-15:
Tested rocket icon on my computer. Windows 11, Chrome 134.0.6998.89.

Got error when COM port was open in another program.
```
Dz2CyDsQ.js:45 11:40:30:626 ERROR [iMeshDevice:SerialConnection] Connect ❌ Failed to execute 'open' on 'SerialPort': Failed to open serial port.
```
I closed the other terminal program, rocket icon identified the board as expected. RAK 4631.

```
Dz2CyDsQ.js:45 11:40:44:650	DEBUG	[iMeshDevice:SerialConnection]	Constructor 🔷 SerialConnection instantiated
Dz2CyDsQ.js:45 11:40:44:676	DEBUG	[iMeshDevice:SerialConnection]	Configure ⚙️ Requesting device configuration
Dz2CyDsQ.js:45 11:40:45:030	INFO	[iMeshDevice:SerialConnection]	HandleFromRadio 📱 Received Node info for this device
Dz2CyDsQ.js:45 11:40:45:032	WARN	[iMeshDevice:SerialConnection]	HandleFromRadio ⚠️ Unhandled payload variant: deviceuiConfig
Dz2CyDsQ.js:45 11:40:45:036	INFO	[iMeshDevice:SerialConnection]	HandleFromRadio 📱 Received Node Info packet for node: 1883245022
Dz2CyDsQ.js:45 11:40:45:037	DEBUG	[iMeshDevice:SerialConnection]	GetMetadata 🏷️ Received metadata packet
Dz2CyDsQ.js:73 Found device onDeviceMetadataPacket Proxy(Object) {hwModel: 9, hwModelSlug: 'RAK4631', platformioTarget: 'rak4631', architecture: 'nrf52840', activelySupported: true, …}
Dz2CyDsQ.js:45 11:40:45:040	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 0
Dz2CyDsQ.js:45 11:40:45:040	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 1
Dz2CyDsQ.js:45 11:40:45:040	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 2
Dz2CyDsQ.js:45 11:40:45:040	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 3
Dz2CyDsQ.js:45 11:40:45:040	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 4
Dz2CyDsQ.js:45 11:40:45:041	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 5
Dz2CyDsQ.js:45 11:40:45:041	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 6
Dz2CyDsQ.js:45 11:40:45:042	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 🔐 Received Channel: 7
Dz2CyDsQ.js:45 11:40:45:044	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: device
Dz2CyDsQ.js:45 11:40:45:046	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: position
Dz2CyDsQ.js:45 11:40:45:049	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: power
Dz2CyDsQ.js:45 11:40:45:052	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: network
Dz2CyDsQ.js:45 11:40:45:053	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: display
Dz2CyDsQ.js:45 11:40:45:055	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: lora
Dz2CyDsQ.js:45 11:40:45:056	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: bluetooth
Dz2CyDsQ.js:45 11:40:45:059	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: security
Dz2CyDsQ.js:45 11:40:45:061	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: sessionkey
Dz2CyDsQ.js:45 11:40:45:061	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Config packet of variant: deviceUi
Dz2CyDsQ.js:45 11:40:45:063	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: mqtt
Dz2CyDsQ.js:45 11:40:45:067	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: serial
Dz2CyDsQ.js:45 11:40:45:068	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: externalNotification
Dz2CyDsQ.js:45 11:40:45:070	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: storeForward
Dz2CyDsQ.js:45 11:40:45:072	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: rangeTest
Dz2CyDsQ.js:45 11:40:45:074	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: telemetry
Dz2CyDsQ.js:45 11:40:45:077	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: cannedMessage
Dz2CyDsQ.js:45 11:40:45:079	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: audio
Dz2CyDsQ.js:45 11:40:45:081	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: remoteHardware
Dz2CyDsQ.js:45 11:40:45:083	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: neighborInfo
Dz2CyDsQ.js:45 11:40:45:085	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: ambientLighting
Dz2CyDsQ.js:45 11:40:45:087	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: detectionSensor
Dz2CyDsQ.js:45 11:40:45:089	TRACE	[iMeshDevice:SerialConnection]	HandleFromRadio 💾 Received Module Config packet of variant: paxcounter
Dz2CyDsQ.js:45 11:40:45:096	WARN	[iMeshDevice:SerialConnection]	HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
Dz2CyDsQ.js:45 11:40:45:099	WARN	[iMeshDevice:SerialConnection]	HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
Dz2CyDsQ.js:45 11:40:45:103	WARN	[iMeshDevice:SerialConnection]	HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
[... cut ...] d variant: fileInfo
Dz2CyDsQ.js:45 11:40:45:110	WARN	[iMeshDevice:SerialConnection]	HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
Dz2CyDsQ.js:45 11:40:45:113	INFO	[iMeshDevice:SerialConnection]	HandleFromRadio ⚙️ Valid config id received from device: 41016316
Dz2CyDsQ.js:73 Checking if remote file exists:  https://raw.githubusercontent.com/meshtastic/meshtastic.github.io/master/firmware-2.5.20.4c97351/firmware-rak4631-2.5.20.4c97351.uf2
Dz2CyDsQ.js:128 Parsing release notes markdown
```

LazyHatGuy, 2025-03-15:
I'm pretty sure I don't have the COM port open on anything else so doubt that's the issue

LazyHatGuy, 2025-03-15:
Forgot to mention my testing was done on Pop-OS and Manjaro Linux

LazyHatGuy, 2025-03-21:
The seems to be fixed in UnGoogled Chomium Version 134.0.6998.88

closed 2025-03-21 by LazyHatGuy