meshtastic/web-flasher #213: Request to Add Support for RAK WisBlock 3312

DanielCao0, 2025-07-15:
[image]

DanielCao0, 2025-09-26:
@rcarteraz @caveman99 The RAK3312 has been released for quite some time. We hope it can be made available on the web-flasher as soon as possible; otherwise, it may affect customer usage. Thank you .

nreisbeck, 2025-10-03:
> [@rcarteraz](https://github.com/rcarteraz) [@caveman99](https://github.com/caveman99) The RAK3312 has been released for quite some time. We hope it can be made available on the web-flasher as soon as possible; otherwise, it may affect customer usage. Thank you .

[image]

It appears to be included in the web flasher now?